graiola/wolf-setup #1: Could not load class "wb_controller/Controller"

itaouil, 2021-11-13:
Ciao Gennaro :),

Thanks for open sourcing this work.

I am thinking to use your package as a way to make AlienGo climb stairs by integrating it with a planner.

I was trying to run the wbc in simulation by following the instructions in your README (through a system installation), but when I launch the package I obtain the following error:

[image]

But the **controller_plugin.xml** seems to be exported in the **package.xml**. Am I missing something?

Best,
Ilyass

graiola, 2021-11-13:
Hi @itaouil, it seems that a debian package was not correctly installed or sourced. Can you check if you have the debian wbc-setup/debs/bionic/advr/cartesian_interface-2.0.0-6d1e2e1-amd64-bionic.deb correctly installed? Also, do you have git-lfs?

itaouil, 2021-11-13:
Hi,

Thanks for the prompt reply.

The debian package **cartesian_interface-2.0.0-6d1e2e1-amd64-bionic.deb** is present in the **wbc-setup** workspace.

And yes, I made sure to install **git-lfs**.

It actually installs the WBC debian packages, although it throws some warnings:

[image]

And similarly for the ADVR debian packages:

[image]

graiola, 2021-11-13:
Can you roscd cartesian_interface?

itaouil, 2021-11-13:
No.

I tried and there is no such package installed.

[image]

graiola, 2021-11-13:
Can you confirm that the pkg is installed with dpkg -l | grep cartesian_interface? Also, could you try echo $ROS_PACKAGE_PATH ?

Thank you!

itaouil, 2021-11-13:
The output of **dpkg -l | grep cartesian_interface**:

[image]

The output of **$ROS_PACKAGE_PATH**:

[image]

graiola, 2021-11-13:
It looks like the xbot distro is not being sourced... try in a new terminal to run `source /opt/xbot/setup.bash` and then launch the wb_controller `roslaunch wb_controller wb_controller_bringup.launch`

itaouil, 2021-11-14:
Yes. Sourcing the xbot setup fixed the issue :).

Thanks for you prompt replies and help.

closed 2021-11-14 by itaouil

itaouil, 2021-12-02:
Hi @graiola,

In the latest push I think you might have forgot to include something while generation the debs:

[image]

graiola, 2021-12-02:
Sorry, I should generate the debians in a docker container instead I am using my computer which sometimes has developments going on... I generated the new debians. Sorry for the inconvenience.

itaouil, 2021-12-02:
No problem at all :).

Thank you!

itaouil, 2021-12-02:
Problem seems to be still present :)

[image]

graiola, 2021-12-02:
Give me few minutes...

graiola, 2021-12-02:
Give it a try :)

itaouil, 2021-12-02:
Error still present :)

[image]

graiola, 2021-12-02:
Ok so I guess my environment is dirty... I would suggest you to use the old debian for the moment... I am going to work on it in the next hours

graiola, 2021-12-03:
I just updated the debian again... hopefully it works this time :)

itaouil, 2021-12-03:
Thanks for the push Gennaro.

While trying to install I receive this:

[image]

Before I try to downgrade the xbotinterface I want to make sure this is the right step? :)

graiola, 2021-12-03:
I just tested the current configuration in the docker image, and it seems to work fine :)

anyway our plan is to move the debians out of git in the next days to make the download process simpler

itaouil, 2021-12-03:
Yes. The docker image works fine.

The docker works fine, but I think the wrong **xbotinterface** was used when you generated the debs for the system installation maybe?

Anyway, I will use the new push with the docker for the moment :)

itaouil, 2021-12-03:
Yes. The docker image works fine :).

There still seems to be some issues with the system install, but I will stick to using the docker image for the moment:

[image]

graiola, 2021-12-03:
Hi @itaouil , we updated the way the debians are stored, now they are on dropbox. If you want, you can try to download the debians with install_dependencies.sh or get_debians.sh. Also, we removed git-lfs and purged the debians from the repo.

If you can give it a try it would be super helpful :)

Thank you!

itaouil, 2021-12-03:
Great!

I'll give it a try in a few hours :)

itaouil, 2021-12-04:
Hi, 

Just gave it a try. I encountered some errors in the installation process:

1) 

**Even when a run the script with sudo**

[image]

which I guess leads to this error:

2) 
[image]

graiola, 2021-12-04:
Hi @itaouil, thank you for the feedback :)

are you using `install_dependencies.sh` to install the debians or do you do it manually?

Regarding the gazebo model: you can add your new model to the `$GAZEBO_MODEL_PATH`, then it should show up when you launch gazebo.
If you are using the docker, you can load your local ros workspace with the model in it with the option `run_docker.sh --local your_ros_ws`, but the container will not have the  `$GAZEBO_MODEL_PATH` updated. You could try to modify line 15 in `fun.cfg` with:

`-e GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/melodic/share/wb_gazebo_resources/models/`

so that the container gets your modified `$GAZEBO_MODEL_PATH`

if it works you can open a PR :)

Regarding the robot model, you can work in a similar way: just put everything in your local ros workspace and run the docker container with the local option. The way the robots are loaded is the following:
There are an `upload.launch` and `world.launch` in `wb_controller/launch` that are called when the controller is spawned. These two launch files are responsible for loading the `robot_description` and `robot_semantic_description` on the ros param server which are the `urdf` and `srdf` files of the robot. So, technically you could imitate what I do with a robot `urdf` and `srdf` (for examples look at `aliengo_description/robots`) and load the new robot with `roslaunch wb_controller wb_controller_bringup robot_name:=new_robot`.
Be careful that you will need also a set of params for the robot (look at `wb_controller/params` to get an idea).

I am going to move the robot params into each respective robot to simplify the loading :)

itaouil, 2021-12-04:
Hi @graiola,

Thanks for the info.

Yes, I am using the script to install the dependencies.

The problem is similar to the one shown in the [link](https://askubuntu.com/questions/1328914/how-to-give-apt-user-permission-to-var-lib-apt-lists/1328915#1328915). I tried to change to permissions before the **install.sh** script is run but with no luck so far.

Also, I noticed that you removed the  **XbotInterface-2.0.0** and now rely only on **XbotInterface-2.1.0**, but the xbot msgs is still at version 2.0.0, while xbot_env at version 1.0.0, could this be related to this error?

[image]

itaouil, 2021-12-04:
Ok.

I managed to install it by **--force-overwrite** the installation of the packages, but then:

[image]

[image]

graiola, 2021-12-04:
In the `debs/advr` folder, there is `uninstall.sh`, can you run it? and then run `install.sh`

graiola, 2021-12-04:
> [image]

ahaha why there are two robots? :):)

itaouil, 2021-12-04:
> In the `debs/advr` folder, there is `uninstall.sh`, can you run it? and then run `install.sh`

[image]

graiola, 2021-12-04:
> > In the `debs/advr` folder, there is `uninstall.sh`, can you run it? and then run `install.sh`
> 
> [image]

it looks like a problem with _apt:

Give a look [here](https://newbedev.com/what-does-this-apt-error-message-download-is-performed-unsandboxed-as-root-mean)

itaouil, 2021-12-04:
> > > In the `debs/advr` folder, there is `uninstall.sh`, can you run it? and then run `install.sh`
> > 
> > 
> > [image]
> 
> it looks like a problem with _apt:
> 
> Give a look [here](https://newbedev.com/what-does-this-apt-error-message-download-is-performed-unsandboxed-as-root-mean)

Adding these two lines 

```
sudo chown -Rv _apt:root $SCRIPTPATH/debs/$UBUNTU/advr/
sudo chmod -Rv 700 $SCRIPTPATH/debs/$UBUNTU/advr/
```

after the **get_debians.sh** call in the **install.sh** script fixes the installation error, but the two robot error spawn persists haha:

[image]

[image]

graiola, 2021-12-04:
how are you running everything? In the docker, outside?

itaouil, 2021-12-04:
No no the docker is fine, and works great.

All these errors are for the system installation only.

graiola, 2021-12-04:
what if you spawn spot? `robot_name:=spot`

graiola, 2021-12-04:
> The docker image works fine. Regarding this, where can I add a new world (under wb_gazebo_resources?) and a new aliengo model (I want to add a sensor plugin to it to have also some perception in the simulator along with the WBC)?
> 

I pushed the modification in `run_docker.sh` about the `$GAZEBO_MODEL_PATH`, feel free to try it

itaouil, 2021-12-05:
> Adding these two lines
>
> ```
> sudo chown -Rv _apt:root $SCRIPTPATH/debs/$UBUNTU/advr/
> sudo chmod -Rv 700 $SCRIPTPATH/debs/$UBUNTU/advr/
> ```
>
> after the **get_debians.sh** call in the **install.sh** [...]

This is what made the controller fail for me in the system installation process:

```
source ~/workspace/code/larvio_gpu/ros_wrapper/devel/setup.bashsource /opt/xbot/setup.bash
```

The sourcing command for **xbot** was added in the same line as the last one and not in a new one by the **install_dependencies.sh** script. Once I added an extra newline all worked fine.

graiola, 2021-12-05:
That's weird because I explicitly append `/opt/xbot/setup.bash` in the bashrc... https://github.com/graiola/wolf-setup/blob/master/install_dependencies.sh#L62

Anyway, I am glad that the issue is fixed now.

Do you still have the problem with the two robots spawning?

itaouil, 2021-12-05:
No everything works fine now :). No double robots party anymore.

itaouil, 2021-12-10:
Hi :),

I changed the name of this issue so it accounts for all errors related to starting the controller.

In the last push the **wolf_description_utils** are not included so it cannot load the models:

[image]

Thought to let you know in case you missed it :)

P.S: If you might want to be notified about this sort of stuff via a different channel or by opening a new issue whenever the error is different. Let me know :)

graiola, 2021-12-10:
I just added the debian :)

itaouil, 2021-12-11:
Yes. Now it works :).

Possible platforms could be:

- Telegram
- Slack
- Discord
- Mattermost

These are the ones that pop in my mind atm.