graiola/wolf-setup #2: High level control without joystick

itaouil, 2021-11-14:
Hi Gennaro :),

In the **README** it is stated that in order to control the robot (I guess both in simulation and real-life?) a joypad is required.

I would be interested to send **twist** messages (i.e. sending the different components for the linear and angular velocity) through ROS. Is there maybe a parameter that could be changed to make this available or would this require changing the source code?

Best,
Ilyass

EnricoMingo, 2021-11-15:
I think there should be a keyboard interface as well. 
@graiola

graiola, 2021-11-15:
Yes, there is a keyboard interface that send twist commands, but it needs further developments. If you explain me what do you need, I can try to implement it.

itaouil, 2021-11-15:
What I am interested in is to have a rostopic called **cmd_vel** that accepts **geometry_msgs/Twist** messages containing the linear and angular velocities which are then passed to the **foothold planner** as references to your **wbc**.

graiola, 2021-11-23:
@itaouil that would be possible :)
I can work on that in the next days

itaouil, 2021-11-23:
Amazing!

Yes. I guess after some further testing in simulation and maybe once the joint violations #6 is solved I will give it a go on the real robot and of course would keep in the loop about that :).

Once again thanks a lot to you and Enrico for all the help! Really appreciated :)

graiola, 2021-11-26:
I resurrected an old branch where there is a twist interface. With the new debian you can publish on the topic `/robot_name/wb_controller/twist` a value between -1.0 and 1.0 for each velocity component, this value represents a scaling value that gets multiplied by the velocity selected in the gui (or in the param file).
To activate the topic use the argument `input_device:=twist` when launching the controller:

`roslaunch wb_controller wb_controller_bringup.launch robot_name:=aliengo input_device:=twist`

Maybe this is not the best way to implement it but I kept it like that to have a similar behavior between the joypad and the twist. If you need a different implementation let me know :)

itaouil, 2021-11-26:
Thanks 👍🏽.

I will give it a try and let you know :).

EnricoMingo, 2021-12-18:
Can we close this?

itaouil, 2021-12-18:
Yes.

Thank you!

closed 2021-12-18 by itaouil

itaouil, 2022-01-05:
Hi @graiola,

To command a clockwise or counter clockwise rotation with the twist message is it a combination of angular.z and almost zero linear.y components?

As angular.z alone only changes the yaw of the base.

graiola, 2022-01-05:
yes you have to send some linear components as well...

itaouil, 2022-01-05:
ok thanks.