graiola/wolf-setup #3: Stair Climbing

itaouil, 2021-11-14:
Hi Gennaro :),

I am interested to use your work for a stair climbing approach with vision targeted to the AlienGo robot.

In the paper it is stated that the wbc might have stability issues when dealing with high inclinations, but I was wondering whether the wbc could potentially deal with stair climbing (16cm steps) or not out of the box (I am yet to try blind stair climbing in simulation as I dont have a joypad yet)?

Best,
Ilyass

EnricoMingo, 2021-11-17:
Dear @itaouil, could you move the video to this discussion, please?

itaouil, 2021-11-17:
Video of blind stairs climbing of aliengo in the simulator originally posted in the discussion #4: https://drive.google.com/file/d/1vKy0tUloRkoicj5D8c86OecrJgv95DCX/view?usp=sharing

EnricoMingo, 2021-11-18:
@itaouil, did you try to address the stairs using the crawl?

itaouil, 2021-11-18:
@EnricoMingo not yet.

I will do further tests today or tomorrow and will update you :).

EnricoMingo, 2021-11-18:
Thanks, @itaouil.
consider that the controller may need to be further tuned (since we support multiple robots we did not fine-tune all of them). May I ask: do you have as well a hardware platform to eventually try your work?

itaouil, 2021-11-18:
Yes.

We have an AlienGo robot here where I could try the controller out possibly, but first I want to make sure that it is robust enough in simulation.

Regarding the fine tuning process, is there any resource/package/software you would recommend to find the best parameters for the robot that could improve robustness?

EnricoMingo, 2021-11-18:
I am sorry but I am not really aware of packages for tuning.

graiola, 2021-11-18:
@itaouil could you send me the stairs model? I am curious to try it as well :)

itaouil, 2021-11-19:
@graiola absolutely :)

Here is the stairs model I am using:

[stairs.zip](https://github.com/graiola/wbc-setup/files/7569069/stairs.zip)

itaouil, 2021-11-22:
Hi @graiola and @EnricoMingo,

I tried the new debs that you pushed for the issue #5 to see if the robot is now able to also climb stairs blindly.

Here is the result https://drive.google.com/file/d/1vFOuyofax-AGkbqvw2kgO0hx1W63Al7I/view?usp=sharing

**Spoiler alert** It managed :). The only thing that I noticed and that I also raised in issue #5 is the joint violations messages that  the wbc throws, which I guess is due to the "weird" joint positions that the robot acquires as soon as it starts moving, but let's keep the discussion about that in the other issue #5.

One final question, did you also include the change in the base height and base attitude based on the slope in the new debs update, or this was something already there before?

EnricoMingo, 2021-11-22:
Dear @itaouil, this is amazing! Thank you for sharing! 
We are working on any aspect of the framework, in particular, concerning the whole-body control stack, we are trying different combinations of tasks and constraints (as well priorities). Our main goal is to try to achieve a stable walk based on reduced feedback data. The base adaptation was not present in the debians before but it was present in another branch. Since we changed the stack quite a lot we decided to put it back.

itaouil, 2021-11-23:
Thanks!

I will close this issue as the robot is now able to climb stairs.

closed 2021-11-23 by itaouil