graiola/wolf-setup #4: WBC failure

itaouil, 2021-11-15:
Hi Gennaro,

I tried to move the robot around with the joystick in the simulator and I noticed that the wbc fails a lot with the robot (in this case AlienGo) starting to fly away in the air.

[image]

And this is the error I get on the terminal:

[image]

Is this due to some parameters having to be changed?

Best,
Ilyass

EnricoMingo, 2021-11-15:
Hi Ilyass, 
from the warnings looks like you are taking a step that is too long (> than 0.5 [m]?)

itaouil, 2021-11-15:
Hi @EnricoMingo,

The thing is that I did not change any parameters, and the base and angular velocities are still set at 0.5 and 0.4 respectively.

Does this mean that the linear and angular velocities have to be further reduced in order for the robot not to take too big of step?

EnricoMingo, 2021-11-15:
Are you using a joystick ?

itaouil, 2021-11-15:
Yes, I am using a joystick.

graiola, 2021-11-15:
@itaouil let me double check that the debians are aligned with the correct code branch. I came back to you asap.

itaouil, 2021-11-15:
Thanks a lot Gennaro :)

graiola, 2021-11-15:
@itaouil could you try to run spot and see if you have the same failures? Thank you :)

itaouil, 2021-11-15:
I tried with all robots now and they all fail. Here I tried spot and I did not even send velocity commands. It was just stumping its feet on the same place and failed.

[image]

[image]

graiola, 2021-11-16:
That's weird... I just udpated the debians, you can try the new ones.

itaouil, 2021-11-17:
Ok thanks :)!

I will try it later today and will update you.

By the way, does it not fail for you when you try?

itaouil, 2021-11-17:
Hi Gennaro :),

I tried the updated debians, and it seems like the behaviour of the robot on flat surfaces in simulation is better than before although it still fails in some occasions though (with the error step size > 0.5).

I also tried to climb stairs blindly with your whole body controller just to see whether it is able to deal with it or not. It did well in the first few steps by then gave up :).

I attached a video to show you what I mean. If you have suggestions on parameters or maybe modifications that I could make to make it more stable let me know :).

Here is the link to the video: https://drive.google.com/file/d/1vKy0tUloRkoicj5D8c86OecrJgv95DCX/view?usp=sharing

I know that you are not actively working on this repo anymore, so thanks a lot for the help you managed to give so far. Really appreciated :).

EnricoMingo, 2021-11-17:
Hi Ilyass,
thank you for the video! I would suggest you to adjust the pose of the base
once you attack the stairs. Probably you just need to move slightly up-back
the base and rotate (pitch) a little on the direction of the slope?
Please let us know how is going.
Best,

Enrico

EnricoMingo, 2021-11-17:
I think this can be closed now. @graiola

itaouil, 2021-11-17:
Sure.

I can move it to the other discussion.

I did some other tests with the new push but I still encountered some failures.

**Case 1: Sudden change in the linear velocity** (https://drive.google.com/file/d/1Gnlj-lxAZ1jkON09iWCssMsvGKuzrtBl/view?usp=sharing)

Here I was playing with the linear velocity and when I suddenly changed it from around 0.7 to around 0.4 the robot failed. Maybe it is due to the sudden change?

**Case 2: Continuous rotation** (https://drive.google.com/file/d/1qBomvBgDhYLUuRjbBL7IwLznpYhWsuU8/view?usp=sharing)

Here I was given as reference a constant angular velocity of 0.4 and the robot lost balance after half rotation.

**Case 3: Step-wise change in the linear velocity** (https://drive.google.com/file/d/1UhCx-SkGlE3kGogm-3B3F9LYJkEccuKW/view?usp=sharing)

Here I was giving increasingly larger velocities but not while the robot was in movement. At 0.7 velocity and after few seconds failed. This test was in order to check if maybe **case 1** failure was indeed due to the sudden change in the velocity but it did not seem the case.

Have you even experienced similar issues before? Or maybe you fine-tuned the parameters in a specific way for your tests? Because it seems to work so well but at sudden moments it just fails with the same error (step larger than 0.5).

EnricoMingo, 2021-11-18:
@graiola you can close.

itaouil, 2021-11-18:
@EnricoMingo I can close it if there are no insights regarding my previous post and the three videos showing some failures on flat surfaces? :)

EnricoMingo, 2021-11-18:
I would close since the videos are not directly related to the issue. If
you mention them in another issue we can close this one and continue on the
other one.

itaouil, 2021-11-18:
Sure.

Will create a new issue then :).

Thanks!

closed 2021-11-18 by itaouil