graiola/wolf-setup #6: Joint-Limits Violations

itaouil, 2021-11-23:
Hi again :),

As started to discuss in #5, there are some joint-limits violations after the new debs update, as shown in the pictures below:

**Starting joint positions (once the solver is on)**

[image]

**After a few steps**:

[image]

**Joints warnings received**:

[image]

graiola, 2021-11-23:
I am going to run some tests in the next days to see what's happening...

graiola, 2021-11-23:
I run some tests and it looks like it is a tuning problem because with spot it works ok.
Try to use these parameters:

[params.txt](https://github.com/graiola/wbc-setup/files/7589612/params.txt)

itaouil, 2021-11-23:
The new parameters helped for the posture :).

I also took another video in case it helps: https://drive.google.com/file/d/1cP9xFqdr8kj8cB3RrAXXkot6jGqZVueP/view?usp=sharing.

In the video you can see the AlienGo limping. I guess this is still due to the joint violations. The limping also happened before the new params, but less frequently I would say.

EnricoMingo, 2021-11-23:
Can you check the velocity references that you are sending? Because it looks like sometimes it takes two steps with the same leg. Maybe you have also to play a little with the parameters of the walking pattern generator.

graiola, 2021-11-23:
Hi @itaouil , we just updated the code to take into account the joint limits into the wbc. It should work better but since the joint limit constraints are working by approximation, it could happen that the controller still throws out some warnings...

Let us know if it improves the doggo behavior! :)

itaouil, 2021-11-25:
Hi,

Sorry for the late reply.

I just tried to test the new debs, but weirdly enough I obtain a similar error as issue #1:

[image]

When I try the previous commit, the same error does not happen.

graiola, 2021-11-25:
Maybe I did a mistake while creating the debians, I am going to generate them again in few minutes.

graiola, 2021-11-25:
@itaouil try again please...

itaouil, 2021-11-25:
Just pulled down the new debs, and the previous issue is solved.

I will do some tests tomorrow finally, and will update you :).

itaouil, 2021-11-26:
I just finished testing the the new debs.

I approached it by sending different velocity commands from 0.1 till 0.9 included with a step of 0.1 for 20 seconds for the following motions:

- Forward
- Right step
- Left step
- Clockwise rotation
- Counter clockwise

for both 6 cm and 15 cm step heights.

Except for some warnings of joint violations and step length > 0.4 and two falls (during a continuous in-place rotation at 0.6 and 0.9) in approximately 40+ minutes of testing it worked remarkably well, and the reactive stairs climbing ability is still working great :)

I will be starting collecting some data in simulation for my project so I will further test it over the next days, but I think the wbc has greatly improved in its stability and robustness since I first tried it, except for rare circumstances like continuous in-place rotation at high velocity (which don't always make the robot fall) and which I will try to investigate in the near future.

Thanks again for open-sourcing this amazing work and the help! :)

closed 2021-11-26 by itaouil

graiola, 2021-11-26:
Hi @itaouil , thank you for testing! You can ignore the warnings because they are generated from an ik module that is now not used anymore... I am going to push a new debian to silence them :)