graiola/wolf-setup #7: Question -- CoM Displacement

itaouil, 2021-11-25:
Hi :),

I remember that the robot in RVIZ used to also move around, instead of being fixed on the place. Is there a parameter I can change to undock it again? 

The reason for it is that I want to extract the absolute CoM displacement every footstep, but for that I need to have the CoM position w.r.t the world frame so that I can compute the actual displacement, or maybe the wbc is already publishing somewhere the CoM displacement?

graiola, 2021-11-26:
Hi @itaouil , you can change in the params the type of estimation for the position, in your case you want to use the ground_truth from gazebo:

` estimation_position_type: ground_truth`

this will give you the position of the robot wrt the world in gazebo.
Let me know if it works :)

itaouil, 2021-11-26:
Hi,

Yes, by changing the **estimation_position_type** the **/aliengo/wb_controller/CoM** topics publishes CoM pose with respect to world.

But I also noticed that even if the **estimation_position_type** is set to **estimated_z** a topic called **/aliengo/ground_truth** is published that additionally also has a non-zero orientation :).

I guess I will use **/aliengo/ground_truth** instead as it has the additional orientation of the body and just keep the **estimation_position_type** set to **estimated_z** for ease of visualization.

graiola, 2021-11-26:
Be aware that the ground_truth represents the position of the robot's base and not its com.
It would be cool to have a proper state estimation one day ;)

itaouil, 2021-11-26:
Ah ok. Did not even realise that :).

The difference in the pose between CoM and base is "minimal" for my purpose so that won't be an issue I think :).

closed 2021-11-26 by itaouil