graiola/wolf-setup #8: Add Sensor to URDF

itaouil, 2021-12-05:
Hi,

I modified the URDF file of the AlienGo robot in order to add a camera, along with the required links and rotations to have correct transformations.

[image]

If this is something you would be interested to include in the next push here is the updated URDF file. If not, maybe someone else in the future might want to do the same and could turn useful :).

```xml
<?xml version="1.0"?>

<robot name="aliengo" xmlns:xacro="http://www.ros.org/wiki/xacro">

     <!-- This argument allows us to load joint sensors that measure the internal wrenches -->
    <xacro:arg name="load_force_sensors" default="false"/>
    <xacro:arg name="task_period" default="0.004"/>
    <xacro:arg name="arm_enabled" default="false"/>
    <xacro:arg name="arm_name" default="kinova"/>
    <xacro:arg name="sensors" default="false"/>
    <xacro:arg name="floating_base_link" default="true"/>

    <xacro:include filename="$(find aliengo_description)/urdfs/const.xacro"/>
    <xacro:include filename="$(find aliengo_description)/urdfs/materials.xacro"/>
    <xacro:include filename="$(find aliengo_description)/urdfs/leg.xacro"/>

    <!-- legs transmission -->
    <xacro:include filename="$(find aliengo_description)/urdfs/leg.transmission.xacro"/>
    <xacro:include filename="$(find aliengo_description)/urdfs/leg.gazebo.xacro"/>

    <!-- generic simulator_gazebo plugins for starting mechanism control, ros time -->
    <xacro:include filename="$(find aliengo_description)/gazebo/gazebo.urdf.xacro">
         <xacro:arg name="task_period" value="$(arg task_period)"/>
    </xacro:include>

    <!-- sensors -->
    <xacro:if value="$(arg sensors)">
        <xacro:include filename="$(find sensors_description)/urdfs/sensors/imu.gazebo.xacro"/>
        <xacro:include filename="$(find sensors_description)/urdfs/sensors/asus_xtion.gazebo.xacro"/>
    </xacro:if>

    <!-- Rollover Protection mode will add an additional stick on the top, use "true" or "false" to switch it. -->
    <xacro:property name="rolloverProtection" value="false"/>

    <!-- Debug mode will hung up the robot, use "true" or "false" to switch it. -->
    <!-- <xacro:if value="$(arg DEBUG)">
        <link name="world"/>
        <joint name="base_static_joint" type="fixed">
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <parent link="world"/>
            <child link="base"/>
        </joint>
    </xacro:if>  -->

    <!-- Floating-base Joint -->
    <xacro:if value="$(arg floating_base_link)">
            <link name="world"/>
            <joint name="floating_base_joint" type="floating">
              <origin rpy="0 0 0" xyz="0 0 0"/>
              <parent link="world"/>
              <child link="trunk"/>
            </joint>
    </xacro:if>

    <link name="trunk">
        <visual>
            <origin rpy="0 0 0" xyz="0 0 0"/>
            <geometry>
                <mesh filename="package://aliengo_description/meshes/trunk.dae" scale="1 1 1"/>
            </geometry>
            <material name="orange"/>
[... 179 lines cut ...]
                imu_name="trunk_imu"
                imu_topic="trunk_imu/data"
                update_rate="250"
                rate_mean="0"
                rate_stddev="0"
                rate_bias_mean="0"
                rate_bias_stddev="0"
                accel_mean="0"
                accel_stddev="0"
                accel_bias_mean="0"
                accel_bias_stddev="0">
            </xacro:imu_sensor>
    </xacro:if>

    <xacro:if value="$(arg sensors)">
            <xacro:asus_xtion_depth_sensor
                link_name="camera_depth_frame"
                frame_name="camera_depth_optical_frame"
                camera_name="camera">
            </xacro:asus_xtion_depth_sensor>
    </xacro:if>

</robot>

```

closed 2021-12-05 by itaouil

graiola, 2021-12-05:
That's great! 

I can easily add that and update the debians, or even better, I am going to put the robot descriptions in a different public submodule...

what do you think?

itaouil, 2021-12-05:
Public submodule sounds good :)

graiola, 2021-12-06:
I just created the submodule, you can find it [here](https://github.com/graiola/wolf_descriptions).
I took the occasion to reorganize a bit the repository but I did not generate the new debians yet.

graiola, 2021-12-06:
Now the debians are updated

graiola, 2021-12-07:
Hey @itaouil , I had to remove the submodule because I realized that I have to merge some branches before making it a separate git entity :)

otherwise the merges are going to be too painful...