graiola/wolf-setup #9: Deterministic first foot movement

itaouil, 2021-12-15:
Hi @graiola,

I noticed that when I send motion commands to the robot when this one is static it can either start its motion with the right or left front foot.

Is this decision based on some criteria? Maybe the foot that is most in the front starts?

Best,
Ilyass

graiola, 2021-12-15:
the order which the feet are scheduled when trotting is the following: LF RH and RF LH and it should remain the same all the time.

itaouil, 2021-12-15:
Yes yes you are right.

I phrased my question wrong I guess.

What is the criteria that defines which set of feet starts the motion starting from an idle position (by pairs I mean LF/RH or RF/LH). Cause from I could see both pairs can start equally. Right?

graiola, 2021-12-15:
The order is cyclic, so it starts with LF RH, then when both feet are in contact the scheduler moves to the next pair RF LH, and it repeats...

itaouil, 2021-12-19:
Thanks for the input! :)

closed 2021-12-19 by itaouil