graiola/wolf-setup #10: Robot Model Bug

itaouil, 2021-12-19:
Hi @graiola,

I noticed that the robot model of the robot in RVIZ **for only** the AlienGo is missing the leg joints.

I am not sure if this is because of the latest pushes or maybe it is a bug on my side. Do you also have the same thing in your workspace?

[image]

[image]

EnricoMingo, 2021-12-19:
Do you have any message in the terminal when running rviz with aliengo?

itaouil, 2021-12-19:
I did not see any specific or new errors/warnings compared to before.

```yaml
logging to /home/ilyass/.ros/log/0c6cc6ac-60c0-11ec-b9aa-1418c37b9864/roslaunch-tama-6851.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/opt/ros/melodic/share/wb_controller/launch/wb_controller_bringup_thesis.launch
[1mstarted roslaunch server http://tama:36647/[0m

SUMMARY
========

PARAMETERS
 * /aliengo/joint_state_publisher/publish_rate: 10000
 * /aliengo/joint_state_publisher/type: joint_state_contr...
 * /aliengo/wb_controller/damp_max: 0.1
 * /aliengo/wb_controller/default_base_angular_velocity: 0.4
 * /aliengo/wb_controller/default_base_linear_velocity: 0.5
 * /aliengo/wb_controller/default_contact_threshold: 30.0
 * /aliengo/wb_controller/default_duty_factor: 0.3
 * /aliengo/wb_controller/default_step_height: 0.065
 * /aliengo/wb_controller/default_swing_frequency: 4.2
 * /aliengo/wb_controller/determinant_max: 0.5
 * /aliengo/wb_controller/estimation_orientation_type: imu_magnetometer
 * /aliengo/wb_controller/estimation_position_type: ground_truth
 * /aliengo/wb_controller/gains/CoM/Kd/x: 50.0
 * /aliengo/wb_controller/gains/CoM/Kd/y: 50.0
 * /aliengo/wb_controller/gains/CoM/Kd/z: 50.0
 * /aliengo/wb_controller/gains/CoM/Kp/x: 200.0
 * /aliengo/wb_controller/gains/CoM/Kp/y: 200.0
 * /aliengo/wb_controller/gains/CoM/Kp/z: 200.0
 * /aliengo/wb_controller/gains/CoM/weight: 1.0
 * /aliengo/wb_controller/gains/Postural/weight: 1.0
 * /aliengo/wb_controller/gains/angular_momentum/weight: 0.1
 * /aliengo/wb_controller/gains/clik_gain: 5.0
 * /aliengo/wb_controller/gains/kd_stance/haa: 5.0
 * /aliengo/wb_controller/gains/kd_stance/hfe: 5.0
 * /aliengo/wb_controller/gains/kd_stance/kfe: 5.0
 * /aliengo/wb_controller/gains/kd_swing/haa: 10.0
 * /aliengo/wb_controller/gains/kd_swing/hfe: 10.0
 * /aliengo/wb_controller/gains/kd_swing/kfe: 10.0
 * /aliengo/wb_controller/gains/kp_stance/haa: 100.0
 * /aliengo/wb_controller/gains/kp_stance/hfe: 100.0
 * /aliengo/wb_controller/gains/kp_stance/kfe: 100.0
 * /aliengo/wb_controller/gains/kp_swing/haa: 100.0
 * /aliengo/wb_controller/gains/kp_swing/hfe: 100.0
 * /aliengo/wb_controller/gains/kp_swing/kfe: 100.0
 * /aliengo/wb_controller/gains/lf_foot/Kd/pitch: 0
 * /aliengo/wb_controller/gains/lf_foot/Kd/roll: 0
 * /aliengo/wb_controller/gains/lf_foot/Kd/x: 15.0
 * /aliengo/wb_controller/gains/lf_foot/Kd/y: 15.0
 * /aliengo/wb_controller/gains/lf_foot/Kd/yaw: 0
 * /aliengo/wb_controller/gains/lf_foot/Kd/z: 30.0
 * /aliengo/wb_controller/gains/lf_foot/Kp/pitch: 0
 * /aliengo/wb_controller/gains/lf_foot/Kp/roll: 0
 * /aliengo/wb_controller/gains/lf_foot/Kp/x: 100.0
 * /aliengo/wb_controller/gains/lf_foot/Kp/y: 100.0
 * /aliengo/wb_controller/gains/lf_foot/Kp/yaw: 0
 * /aliengo/wb_controller/gains/lf_foot/Kp/z: 1000.0
 * /aliengo/wb_controller/gains/lf_foot/weight: 100.0
 * /aliengo/wb_controller/gains/lf_haa_joint/d: 5.0
 * /aliengo/wb_controller/gains/lf_haa_joint/i: 0.0
 * /aliengo/wb_controller/gains/lf_haa_joint/p: 100.0
 * /aliengo/wb_controller/gains/lf_hfe_joint/d: 8.0
 * /aliengo/wb_controller/gains/lf_hfe_joint/i: 0.0
 * /aliengo/wb_controller/gains/lf_hfe_joint/p: 300.0
 * /aliengo/wb_controller/gains/lf_kfe_joint/d: 8.0
 * /aliengo/wb_controller/gains/lf_kfe_joint/i: 0.0
 * /aliengo/wb_controller/gains/lf_kfe_joint/p: 300.0
[... 292 lines cut ...]
[0m[1m[info] [0mPilerHelper: expanding to 24 x 1 
[0m[1m[info] [0mPilerHelper: expanding to 42 x 30 
[0m[1m[info] [0mPilerHelper: expanding to 42 x 1 
[0m[1m[info] [0mPilerHelper: expanding to 42 x 1 
[0m[1m[info] [0mPilerHelper: expanding to 62 x 30 
[0m[1m[info] [0mPilerHelper: expanding to 62 x 1 
[0m[1m[info] [0mPilerHelper: expanding to 62 x 1 
[0m[1m[info] [0m#USING BACK-END @LEVEL 0: qpOASES
[0m[1m[info] [0mPilerHelper: expanding to 62 x 30 
[0m[1m[info] [0mPilerHelper: expanding to 62 x 1 
[0m[1m[info] [0mPilerHelper: expanding to 62 x 1 
[0m[1m[32m[ok  ] [0m[32mOpenSotBackEndQPOases so library found! [0m

[1m[info] [0mPROBLEM 0 ID: lf_foot_0to2+lh_foot_0to2+rf_foot_0to2+rh_foot_0to2+waistRPY_3to5+angular_momentum+CoM 
[0m[1m[info] [0mCONSTRAINTS ID: AGGR_AGGR_wrenches_limits+torque_limits+friction_cones 
[0m[1m[info] [0m    # OF CONSTRAINTS: 62 
[0m[1m[info] [0mBOUNDS ID: AGGR_wrenches_limits+torque_limits+friction_cones 
[0m[1m[info] [0m    # OF BOUNDS: 0 
[0m[1m[info] [0m# OF VARIABLES: 30 
[0m[1m[info] [0meps Regularisation factor: 2.221000e-13 
[0m[1m[info] [0mqpOASES # OF CONSTRAINTS: 62
[0m[1m[info] [0mqpOASES # OF VARIABLES: 30
[0m[1m[info] [0m
[0m
```

graiola, 2021-12-19:
I saw the same visualization problem on another computer, but it doesn't happen on mine... I would say that it is a graphic glitch related to the meshes. Maybe changing the meshes could help...? You can try to open the lowerleg meshes with another program on your computer and see if you can correctly visualize them.

itaouil, 2021-12-19:
Yes. I am actually using a new PC.

I tried to open the **calf.dae** mesh with meshlab and it crashes every single time. While for the other meshes everything is correctly visualized.

I also tried to replace the meshes with the other ones from a unitree robotics github repo but the result is still the same.

graiola, 2021-12-19:
It has never been an issue on my computer. Anyway it is only a graphic glitch, it does not affect the simulation or the control (and it doesn't affect the gazebo visualization). It could be related to the drivers... .

itaouil, 2021-12-20:
I see.

Yes true. I was able to work with the controller no problem.

If I find a solution in the future will make sure to update this issue.

Thanks :)

P.S: This might be my last issue of the year haha. Therefore thanks for your all your help so far and happy holidays :)

closed 2021-12-20 by itaouil

graiola, 2021-12-23:
Hi @itaouil , I made public again the descriptions, you can find them here:

https://github.com/graiola/wolf_descriptions

feel free to open PRs :)

itaouil, 2021-12-23:
Nice!

Thanks :)