graiola/wolf-setup #11: Velocity commands not executed correctly in ground_truth estimation position type

itaouil, 2021-12-20:
Hi,

It seems like the behavior of the controller changes when I set the **estimation_position_type** in the param to **ground_truth**.

In particular, when I set a high velocity (say for example 0.9) and switch back to 0.2 for example the robot does not actually execute that lower velocity, but instead it seems to stomp on the spot, as if the velocity set is 0.0.

Here is a small video demonstrating the behavior.

https://user-images.githubusercontent.com/17275225/146845563-9ab8392b-3e35-48a0-8275-814f82f57765.mp4

**The same behavior does not happen when the estimation position type is set to estimated_z btw....**.

Any idea on what is causing such an issue? Could it be something related to the IMU sensor?

graiola, 2021-12-21:
Hi @itaouil , today I am travelling back home. I am going to give a look at the issue asap :)

itaouil, 2021-12-21:
Thanks :)

Safe travel!

graiola, 2021-12-21:
Hi @itaouil , the video doesn't show up. it says that can not be played because the file is corrupted.

itaouil, 2021-12-22:
> Hi @itaouil , the video doesn't show up, it says that can not be played because the file is corrupted.

I uploaded it to drive :)

https://drive.google.com/file/d/1iK-sqa3j57DMBnQW1WlYdTuaUH-kaQ-Y/view?usp=sharing

graiola, 2021-12-22:
It seems there is a bug when using the ground_truth. Did you have this problem before?

itaouil, 2021-12-22:
I actually never tried the **ground_truth** before. I always used the **estimated_z**.

graiola, 2021-12-22:
I think i solved the bug, I am building the debians now....

itaouil, 2021-12-22:
I will try it now and update you asap!

itaouil, 2021-12-22:
Hi,

I gave it a try. It seems that the bug is solved now :).

While trying I noticed couple of behaviors that still persist. For example the shuffling of the support feet in the back (along with some weird posture)

Here is a couple videos demonstrating it (especially the second video link):

https://drive.google.com/file/d/1uqx_RToWp3JJriSPXZY_aZ19Pkbo66x9/view?usp=sharing
https://drive.google.com/file/d/1sNlOdC7eqapSEZWqw2YHsWuSWsv3QjRr/view?usp=sharing

graiola, 2021-12-22:
does it happen when using the estimation?

itaouil, 2021-12-22:
Yes...

Here is a video showcasing the same sort of shuffling behavior while using **estimated_z**:

https://drive.google.com/file/d/1iXTlTBDZdoICsMvYvvEkjpKiik99PAsX/view?usp=sharing

But I remember it quite distinctly not being the case before though.

graiola, 2021-12-22:
Tomorrow I am going to investigate...

graiola, 2021-12-23:
I made a quick revert to an older commit and generated the debians. You can try and see if it solves the issue on your side.
On my computer is working fine (note that I am using a joypad). Let me know :)

itaouil, 2021-12-23:
The shuffling still persists, and the climb up with the stairs still fails with the same slope as the video I posted some time ago.

Unfortunately I cannot even cross-check with my other PC which has the previous old version of the debians whether the same behavior was there back then (although I am 90% sure the shuffling was not as present as right now if not at all) :).

But I will be able to do so on the 1st of January once I am back in Germany :).

itaouil, 2021-12-23:
Anyway, the original problem was solved so I will close this issue.

Thanks for your help even during the holidays. Much appreciated :).

closed 2021-12-23 by itaouil