graiola/wolf-setup #12: Visualize same height in RVIZ as in Gazebo

itaouil, 2022-01-03:
Hi Gennaro,

Even though I set up the **estimation_position_type** to **ground_truth** I just realized that the height of the robot from the ground floor in RVIZ is not the same as the one in Gazebo as shown in the picture below;

[image]

The "more of less correct height" (I know it is the trunk height) of the CoM is available in the **aliengo/ground_truth** topic, but I am not sure how to change the settings so that I can have the correct height also on RVIZ (maybe using the aliengo/ground_truth topic as the odometry topic). Do you have any suggestions on how to change this? :)

graiola, 2022-01-03:
Hi @itaouil, I patched the ground truth visualization, now it should be correct with the new debians. An important thing to remember is that the ground truth is supposed to be used for debugging only and not to run the robot because it represents the robot pose in gazebo which doesn't have any real meaning :)

graiola, 2022-01-03:
Also, if you want to add the stairs to WoLF, you can check out this new repo and open a PR:

https://github.com/graiola/wolf_gazebo_resources

:)

itaouil, 2022-01-03:
> Hi @itaouil, I patched the ground truth visualization, now it should be correct with the new debians. An important thing to remember is that the ground truth is supposed to be used for debugging [...]

Thanks a lot! It works great now:

[image]

Yes. I am only using it for development in simulation, as I don't want to fiddle around with getting to work a state estimator there. I was not planning to use it with the real robot if this is what you meant :).