graiola/wolf-setup #13: Robot starting on the ground

graiola, 2022-01-04:
I am preparing a major release where the robot starts directly from the ground.
@itaouil could you be interested in testing this release? Also, are you using the ROS topic to control the robot? Because I am planning to create some ROS services to perform various actions like resetting the base or standing up etc... .

itaouil, 2022-01-05:
Hi,

Yes, happy to try it out in simulation (if this is what you mean) :).

At the moment I am controlling the robot using the joy commands and velocities that obtain from my planner (without the joystick) but I am planning to change this and start controlling it using the twist interface as it is less cumbersome. But yes I am using ROS topics.

One more thing. Given I am using your controller for my thesis work, would it be possible to keep a clean branch with the current debians so that in case the new release messes something up I still have a backup to go to for my work?

graiola, 2022-01-05:
I updated the debians in bionic with the new stand up and stand down procedure.

itaouil, 2022-01-05:
> I updated the debians in bionic with the new stand up and stand down procedure.

Amazing!

I will give it a try and update you tomorrow or so :)

itaouil, 2022-01-11:
Hi @graiola,

I am sorry for the delay. I have got sidetracked with some stuff...

I did not forget about it though :).

graiola, 2022-01-11:
np :)

elpimous, 2022-01-20:
I'd like it too !

closed 2022-01-31 by graiola