graiola/wolf-setup #17: CREATE ROS SERVICE FOR SEPARATE LEGS AND BODY

elpimous, 2022-01-20:
In a near futur, will you create a ros topic to move isolated foot, or move/rotate body only ?

Working on a python choregraphie app, with sliders (calling robot parts moving), a music, and sound waveform with metronome for tempo, rec instant position, and player.  Idea is to record a music choregraphie, but need to be able to move separate legs, and move/rotate body. Well, use my robot like a puppet

It would be very cool, Thanks
Vincent

graiola, 2022-01-31:
The idea would be to create a plugin system to load different behaviors in the controller. Right now the SoT works only in conjunction with the walking pattern generator presented in the paper.

elpimous, 2022-02-03:
Humm, it was quite easier, in my head..
Next TODO (in my dreams)

closed 2022-02-22 by graiola