graiola/wolf-setup #19: Can't access wolf_controller under rqt_reconfigure, with new xacro robot

elpimous, 2022-01-28:
Hello the team, i made the ylo2 package, copying your aliengo_description one
robot is ok under rviz (sizes, rotations, inertia)
Don't see any errors in terminal : https://pastebin.com/SzhDDbvU
https://ibb.co/ByQbhSm
the soft launchs ok, i can see gazebo params in rqt_reconfigure, but i don't see the ylo2/wolf_controller inside it, even with refreshs

Could it be a startup pose positions problem ? Where could i search for errors ?

graiola, 2022-01-28:
Hi @elpimous , there is an error with the ros controller manager, `Controller Spawner couldn't find the expected controller_manager ROS interface.` 
Do you have the latest packages installed? If not, please run the installation again or get them with the script `wolf-setup/support/get_debians.sh`, you will find them in the folder `wolf-setup/debs/devel/`.

elpimous, 2022-01-28:
Hey, Gennaro, what a quick answer... 
Your new install.sh is nice (logo). Well, installed, ok.
Same problem. It could be the fault of my initial pose... Do you have an imu check before starting the wolf controller ?

elpimous, 2022-01-28:
[ylo2_description.zip](https://github.com/graiola/wolf-setup/files/7959179/ylo2_description.zip)
Here is my robot_description.... if you want to test..

graiola, 2022-01-28:
Yes, I would be very happy to give it a spin asap :)
Also, feel free to open a pull request on https://github.com/graiola/wolf_descriptions with the new robot, so that it would be easier for both of us to work on it.

elpimous, 2022-01-28:
I'll do it for sure..
thanks

graiola, 2022-01-28:
Hey @elpimous , I fixed the issue, there was an `aliengo` in the urdf. I was able to spawn the controller and to move the robot around but the controller needs some tuning. You can get the fix in the `ylo2` branch https://github.com/graiola/wolf_descriptions/tree/ylo2

elpimous, 2022-01-29:
Very good job, Gennaro... Working on it.
Thank you so much.
Vicent

closed 2022-01-29 by elpimous

elpimous, 2022-01-29:
just sent a pull request (first time for me !) Hope it worked