graiola/wolf-setup #26: working errors on ubuntu20.04/noetic

elpimous, 2022-02-20:
hey guys, working on Ylo2robot, with new ubuntu20.04 working Noetic.
my native python is 3.8
```
/usr/bin/env: «python»: Aucun fichier ou dossier de ce type
[goto0-5] process has died [pid 65547, exit code 127, cmd /home/ylo2/catkin_ws/src/wolf_descriptions/wolf_description_utils/scripts/go0
```
installed python, and some python imports fail !
```
Traceback (most recent call last):
  File "/home/ylo2/catkin_ws/src/wolf_descriptions/wolf_description_utils/scripts/go0", line 3, in <module>
    import rospy, sys, os, time, tf
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/noetic/lib/python3/dist-packages/std_msgs/msg/_Bool.py", line 6, in <module>
    import genpy
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/noetic/lib/python3/dist-packages/genpy/message.py", line 48, in <module>
    import yaml
ImportError: No module named yaml
[goto0-5] process has died [pid 1614, exit code 1, cmd /home/ylo2/catkin_ws/src/wolf_descriptions/wolf_description_utils/scripts/go0
```
it seems that i cant install yaml (conflits with pip)

EnricoMingo, 2022-02-20:
Hi!
Can you try

 sudo apt-get install python3-yaml
?

> Il giorno 20 feb 2022, alle ore 21:24, FOUCAULT Vincent ***@***.***> ha scritto:
> 
> ImportError: No module named yaml

EnricoMingo, 2022-02-20:
Or first purge and then reinstall 

> Il giorno 20 feb 2022, alle ore 21:31, Enrico Mingo ***@***.***> ha scritto:
> 
> ﻿
> Hi!
> Can you try
> 
>  sudo apt-get install python3-yaml
> ?
> 
>>> Il giorno 20 feb 2022, alle ore 21:24, FOUCAULT Vincent ***@***.***> ha scritto:
>>> 
>> ImportError: No module named yaml

elpimous, 2022-02-20:
python3-yaml est déjà la version la plus récente (5.3.1-1ubuntu0.1)
already newest version.
You want me to purge the full install ? I'll have same install right after !?!

elpimous, 2022-02-20:
seems looking for yaml for python2 ?!

EnricoMingo, 2022-02-20:
Nope, not purge everything.
If you run ipython or a python shell can you import yaml?

> Il giorno 20 feb 2022, alle ore 21:31, Enrico Mingo ***@***.***> ha scritto:
> 
> ﻿
> Hi!
> Can you try
> 
>  sudo apt-get install python3-yaml
> ?
> 
>>> Il giorno 20 feb 2022, alle ore 21:24, FOUCAULT Vincent ***@***.***> ha scritto:
>>> 
>> ImportError: No module named yaml

elpimous, 2022-02-20:
ImportError: No module named yaml
no

EnricoMingo, 2022-02-20:
I would say you are missing the package…can you locate it in your machine?

graiola, 2022-02-20:
With ubuntu 20.04 you should be using python3. I think something got wrong with your python environment :/

elpimous, 2022-02-22:
hello the team, i saw that i missed the following command after my ROS NOETIC environment installation : 
$ `catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3`
after that, it was ok.

closed 2022-02-22 by elpimous