graiola/wolf-setup #29: libmoveit_robot_model.so updated

itaouil, 2022-03-21:
Hi Gennaro,

I updated my system and the **libmoveit_robot_model.so.1.0.9** library that the controller depends on got updated to 1.0.10.

Would you be able to generate a new version for this update?

Best,
Ilyass

graiola, 2022-03-21:
Hi Ilyass,

I think the debians in the devel folder are already updated.
Can you try them?

Gennaro

graiola, 2022-03-21:
Could you tell me which version of 'wolf_controller' you are using?

itaouil, 2022-03-21:
[image]

graiola, 2022-03-21:
I started the to regenerate the debians but it is going to take some time unfortunately... They should be updated by this evening.

itaouil, 2022-03-21:
Ok thanks :)

graiola, 2022-03-21:
@itaouil the devel debians should be updated. Can you test it? Thanks.

closed 2022-03-21 by itaouil