graiola/wolf-setup #34: missing packages on wolf setup install, on noetic

elpimous, 2022-09-01:
./install.sh on noetic :+1: 

dpkg: des problèmes de dépendances empêchent la configuration de ros-noetic-wolf-controller :
 ros-noetic-wolf-controller dépend de modelinterfacerbdl (= 1.1.10) ; cependant :
  **Le paquet modelinterfacerbdl n'est pas installé.**
 ros-noetic-wolf-controller dépend de xbotinterface (= 2.1.0-1fe46dc-amd64-focal) ; cependant :
  **Le paquet xbotinterface n'est pas installé.

dpkg: erreur de traitement du paquet ros-noetic-wolf-controller (--install) :**
 problèmes de dépendances - laissé non configuré
Paramétrage de ros-noetic-wolf-description-utils (0.0.1-0focal) ...
Paramétrage de ros-noetic-wolf-exploration (0.0.1-0focal) ...
Paramétrage de ros-noetic-wolf-gazebo-resources (0.0.1-0focal) ...
Paramétrage de ros-noetic-wolf-hardware-interface (0.0.1-0focal) ...
**dpkg: des problèmes de dépendances empêchent la configuration de ros-noetic-wolf-navigation-utils :
 ros-noetic-wolf-navigation-utils dépend de ros-noetic-wolf-controller ; cependant :
 Le paquet ros-noetic-wolf-controller n'est pas encore configuré.**

dpkg: erreur de traitement du paquet ros-noetic-wolf-navigation-utils (--install) :
 problèmes de dépendances - laissé non configuré
Paramétrage de ros-noetic-wolf-rviz-plugin (0.0.1-0focal) ...
Paramétrage de ros-noetic-wolf-waypoints (0.0.1-0focal) ...
Paramétrage de ros-noetic-ylo2-description (0.0.1-0focal) ...
Paramétrage de ros-noetic-anymalc-description (1.0.2-0focal) ...
Paramétrage de ros-noetic-spot-description (0.0.1-0focal) ...
Paramétrage de ros-noetic-wolf-gazebo-interface (0.0.1-0focal) ...
**Des erreurs ont été rencontrées pendant l'exécution :
 ros-noetic-wolf-controller
 ros-noetic-wolf-navigation-utils**

Any idea ??

elpimous, 2022-09-01:
worked with "git checkout xbot_apt_server" first

graiola, 2022-09-01:
weird, it should work on the master branch as well... I am going to investigate that. Thank you for reporting the issue :)

elpimous, 2022-09-01:
you're welcome..
but this error occur, even with a `source ~/.bashrc`
```
[ERROR] [1662061215.418513830]: Could not load class 'wolf_controller/Controller': Failed to load library /opt/ros/noetic/lib/libwolf_controller.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libmatlogger2.so.1.3.1: cannot open shared object file: No such file or directory)
[ERROR] [1662061215.418580017]: Could not load controller 'wolf_controller' because controller type 'wolf_controller/Controller' does not exist.
[ERROR] [1662061215.418608335]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1662061216.419894, 0.000000]: Failed to load wolf_controller
```

graiola, 2022-09-01:
Because it is not supposed to work with the xbot_apt_server branch.

graiola, 2022-09-05:
Can you try again? I just updated the debians and the docker images, and they are working fine for me. Thank you :)

elpimous, 2022-09-06:
humm, errors ?!!

https://pastebin.com/ZUfrivuA

graiola, 2022-09-06:
I think the problem is caused by the xbot apt list. Can you remove the xbot source from your apt list and try to install again?

elpimous, 2022-09-06:
removed.
same error : 
https://pastebin.com/dZaQkjfA

elpimous, 2022-09-06:
removed.
same error : https://pastebin.com/XZUYVFKT

graiola, 2022-09-06:
try to run the `uninstall.sh` script in the `debs/advr` folder and install again

elpimous, 2022-09-06:
humm. removed from apt,
removed any xbot_*
reinstall ok
under catkin_ws,  `catkin_make`, then `source ~/.bashrc`, then `roslaunch wolf_controller wolf_controller_bringup.launch`
can't find lib : Failed to load library /opt/ros/noetic/lib/libwolf_controller.so

the error : https://pastebin.com/Fts4uhJM

my ~/.bashrc : 

`...`

elpimous, 2022-09-06:
try to run the uninstall.sh script in the debs/advr folder and install again   working !!
Thanks Gennaro.

closed 2022-09-06 by elpimous