graiola/wolf-setup #35: wolf_controller ros package missing

SOHAILMBS, 2022-09-14:
Hi,

I am trying to build this package from the source but the controller package is missing which contains the binaries. I want to test the package with the GO1 robot and possibly on the real robot as well.

graiola, 2022-09-14:
Hello,

The controller is distributed only as a debian at the moment. The rest of the code instead is open-source. If you want to use GO1, you can give a look at the `wolf_aliengo_interface` package. For any other question or issue I am more than happy to help :)

SOHAILMBS, 2022-09-19:
Pardon the late reply. Thank you for replying. I was hoping that perhaps binaries were available to build it without the docker. Anyway thank you for your time!

closed 2022-09-19 by SOHAILMBS