graiola/wolf-setup #37: Simulation platform agnostic compatibility ?

Astrabrat, 2023-02-23:
Hey,

I highly appreciate the repo's open-source availability to simulate quadrupeds. However, it looks like the repo is currently integrated with Gazebo. Have you given a thought of having the controller be independent of any platforms as it looks like it is difficult to launch the controller without launching gazebo too. Can you suggest a way around this, if you are aware of it.  Also, is it possible to add a list of topics that the controller requires to function as intended to understand the topic mapping around the architecture. 

Great work by the team!

graiola, 2023-06-22:
Hi, yes it is possible to launch the controller without gazebo. In this case, you need to create your specific hardware interface. You can give a look at https://github.com/graiola/wolf_aliengo_interface

Astrabrat, 2023-11-23:
Thank you. Closing this issue.

closed 2023-11-23 by Astrabrat