graiola/wolf-setup #38: Poco exception = libCartesianInterface.so

jianwei-liu93, 2023-05-23:
Hi, 

I'm getting an "libCartesianInterface.so" error with the wolf_controller, any ideas?

[ERROR] [1684869442.020911803, 0.210000000]: Could not load class 'wolf_controller/Controller': Failed to load library /opt/ros/noetic/lib/libwolf_controller.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libCartesianInterface.so: cannot open shared object file: No such file or directory)

jianwei-liu93, 2023-05-24:
Found the issue, /opt/xbot/setup.sh was not sourced for some reason, worked after sourcing

closed 2023-05-24 by jianwei-liu93

graiola, 2023-05-25:
sorry for the late reply, I am glad that the issue is solved :)