graiola/wolf-setup #39: (Poco exception = libsrdfdom.so.0.6.3: cannot open shared object file: No such file or directory) and (ros-noetic-nimbro-service-transport depends on libqt4-dev:   Package libqt4-dev is not installed.)

Astrabrat, 2023-06-22:
Hello, 

Once we ran `./install.sh`,we face the following two issues :

1. `Failed to create robot simulation interface loader: Failed to load library /opt/ros/noetic/lib/libwolf_hw_sim.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libsrdfdom.so.0.6.3: cannot open shared object file: No such file or directory)
`
This looks like the wolf_setup requires xxx.0.6.3 but recently the srdfdom package got updated to 0.6.4. 
 
2. `dependency problems prevent configuration of ros-noetic-nimbro-service-transport:
 ros-noetic-nimbro-service-transport depends on libqt4-dev; however:
  Package libqt4-dev is not installed.
 ros-noetic-nimbro-service-transport depends on qt4-qmake; however:
  Package qt4-qmake is not installed.
`
Since qt4 has been depreciated and  I see the `sys_deps_list.txt` call for qt5, this error propped up only when we ran the recent `./install.sh` command. I tried commenting out  `ros-noetic-nimbro-xx` from `wolf_deps_list.txt` and `wolf.yaml` inside `wolf-setup/config/` folder, but kept coming across the error.

Could you suggest any workarounds for the issues ?

graiola, 2023-06-23:
it should be solved now with the new debian packages

Astrabrat, 2023-06-26:
Thank you for such a quick resolution. It is working as expected. closing this issue.

closed 2023-06-26 by Astrabrat